TKJElectronics/KalmanFilter #4: tag 1.0.1 has library.properties with version 1.0.0

ffissore, 2015-04-15:
Thus leading to a double entry in http://downloads.arduino.cc/libraries/library_index.json

ffissore, 2015-04-15:
See https://github.com/arduino/Arduino/issues/2983

Lauszus, 2015-04-16:
@ffissore hmm it doesn't look like it removes the duplicate automatically: http://downloads.arduino.cc/libraries/library_index.json. Also the version parameter isn't updated? Note how the url is updated correctly, but the version parameter is still 1.0.0.

ffissore, 2015-04-16:
I guess it's because it hasn't used the new tag but it's still stuck with the older. I'm out of the office until next wednesday.

ffissore, 2015-04-16:
In the meanwhile, I'll reopen https://github.com/arduino/Arduino/issues/2983

Lauszus, 2015-04-16:
Okay thanks.

commit a09adafb (2015-04-15): Updated version

Fixes #4
closed 2015-04-28 by Lauszus