TKJElectronics/KalmanFilter #5: general kalman library code, not targeting to IMU sensors

shiftleftplusone, 2016-07-17:
hi,
as I have observed, your lib is called "Kalman.h" but there is no general kalman library code,  instead just an example how to target a MPU IMU sensor.
I'm curious how to use this Kalman lib also for gerneral purposes, i.e. fusioning sensors like GPS and odometry and/or some IR distance sensors.
So what would be needed is a guide (and/or well commented example code) how to initialize the Kalman filter by sensor values and perhaps their statistic properties (e.g., range, standard deviation), then just assign the values to the filter and then let the filter work.

How can that be done ?

Bomanaziba, 2017-02-03:
hi, guys I tried working with Kalman filter library and I had an error 

Below are details

Arduino: 1.8.1 (Windows 10), Board: "Intel® Galileo"

In file included from C:\Users\OGONI\Documents\Arduino\MPU6050\MPU6050.ino:41:0:

C:\Users\OGONI\AppData\Local\Arduino15\packages\Intel\hardware\i586\1.6.7+1.0\libraries\Wire\src/Wire.h: In function 'void setup()':

C:\Users\OGONI\AppData\Local\Arduino15\packages\Intel\hardware\i586\1.6.7+1.0\libraries\Wire\src/Wire.h:90:39: error: local class 'class setup()::TwoWire' shall not have static data member 'const uint32_t setup()::TwoWire::RECV_TIMEOUT' [-fpermissive]

C:\Users\OGONI\AppData\Local\Arduino15\packages\Intel\hardware\i586\1.6.7+1.0\libraries\Wire\src/Wire.h:91:39: error: local class 'class setup()::TwoWire' shall not have static data member 'const uint32_t setup()::TwoWire::XMIT_TIMEOUT' [-fpermissive]

MPU6050:45: error: 'class setup()::TwoWire' has no member named 'setClock'

C:\Users\OGONI\Documents\Arduino\MPU6050\I2C.ino: In function 'uint8_t i2cWrite(uint8_t, uint8_t*, uint8_t, bool)':

I2C:26: error: 'Wire' was not declared in this scope

C:\Users\OGONI\Documents\Arduino\MPU6050\I2C.ino: In function 'uint8_t i2cRead(uint8_t, uint8_t*, uint8_t)':

I2C:39: error: 'Wire' was not declared in this scope

Multiple libraries were found for "Wire.h"
 Used: C:\Users\OGONI\AppData\Local\Arduino15\packages\Intel\hardware\i586\1.6.7+1.0\libraries\Wire
 Not used: C:\Users\OGONI\AppData\Local\Arduino15\packages\Intel\hardware\i586\1.6.7+1.0\libraries\Servo
exit status 1
'class setup()::TwoWire' has no member named 'setClock'

I'll be grateful for any solution, its of dire importance
thank you

shiftleftplusone, 2017-02-03:
please file your own issue!

Lauszus, 2018-06-23:
No this library is not meant as a "general" Kalman filter. You can read more about it here: <http://blog.tkjelectronics.dk/2012/09/a-practical-approach-to-kalman-filter-and-how-to-implement-it>.

I might release a more general library at some point, but for now I am too busy with other things. But please note that there are not such thing as a general Kalman filter that works with all applications,  the engineer will always have to define the state matrices etc.

When it comes to tuning the filter, then I mostly tune it offline on sampled data and then analyse it in either Matlab or Python.

If you really want to know more about the Kalman filters, then I would recommend the following books: [Linear Systems Control](https://www.springer.com/gb/book/9783540784852) and [Aided Navigation](https://www.amazon.com/Aided-Navigation-High-Rate-Sensors/dp/0071493298).

And yes I agree that it is probably not the best name, but I didn't really expect it to get so popular when I first created the library. Anyway I am not going to rename the library just because you find it confusing.

closed 2018-06-23 by Lauszus