TKJElectronics/KalmanFilter #9: No waiting needed after Wire.requestFrom().

Koepel, 2017-08-08:
In the file "examples/MPU6050/I2C.ino" there is a complicated way with polling Wire.available() and timeouts and micros() after the call to Wire.requestFrom(). That can be removed, there is nothing to wait for.

When the Wire.requestFrom() returns, the I2C transaction has completely finished and the received data is waiting in a buffer in the Wire library.

The preceding Wire.endTransmission() is already tested for a I2C bus error. If you want, you can add a test for Wire.requestFrom() as well.

```
  Wire.requestFrom(IMUAddress, nbytes, (uint8_t)true); // Send a repeated start and then release the bus after reading
  if(Wire.available() != nbytes)
  {
    Serial.println(F("i2cRead bus error"));
    return 5; // This error value is not already taken by endTransmission
  }
  else {
    for (uint8_t i = 0; i < nbytes; i++) {
      data[i] = Wire.read();
    }
  }
```

Lauszus, 2018-06-23:
Since this does not have real benefits and the code in the library have been proven to work with multiple different kind of microcontrollers, then I see no real benefit to changing it. If you really want to change it, then you can submit a PR and do some testing.

closed 2018-06-23 by Lauszus

Koepel, 2018-06-23:
I don't want to make a Pull Request and test it. Sorry. Why would I test a bug against not a bug ? I think it is not okay to use the Wire library in the wrong way in the first place.

When others use your library in a balancing robot, they often need a high sample rate. The unnecessary extra I2C bus transaction caused by the Wire.endTransmission() after the Wire.requestFrom() makes it slower. Do you mind if I tell others about it ?

Lauszus, 2018-06-23:
The reason why it was implemented that way was because the ```readBytes``` function was not available when I wrote the code, as I also wrote here: https://github.com/Lauszus/BalancingRobotFullSize/issues/2. Btw opening up several issues with the same title and text and sending me emails doesn't help in getting my attention - it's just annoying.

If you really believe it such a huge issue, why don't you make a pull request and prove it? Have you actually measured the performance difference between the two codes? I suspect that the improvement are negligible, thus won't have any real life impact. But please prove me wrong.

If you really care about high sample rate, then you should use the SPI interface if the sensor supports it.

Koepel, 2018-06-23:
Does this mean that you still don't know how to use the Wire library ?

If have read the library source code what a Wire.endTransmission() does on its own (that is: after a Wire.requestFrom) and I have confirmed the unnecessary bus transaction with a logic analyzer. I don't have a screendump of that. I really don't understand what I have to prove. I only want to tell you that you use the Wire library in the wrong way. It is up to you what to do with that.

Lauszus, 2018-06-23:
@Koepel no it does not. It means that I don't have the time to change and test it. My concern is that the Wire implementation is different for all the different Arduino ports out there. For instance have you verified this on AVR, Arduino Due, Teensy, ESP32, ESP8622 etc?

Koepel, 2018-06-23:
Only the source code. The Teensy has other functions as well, but they have other names. The Teensy functions that are compatible with the Wire library are also according to the Arduino Wire documentation. The ESP has a few extra functions but is also according to the Arduino Wire documentation.

I think we are at an impasse. To be compatible with other Wire libraries, it is important to use the Wire library correctly. That means that using the Wire library in the wrong way will make it less compatible. I don't understand why I would have to verify that using the Wire library functions in the correct way is better.