TKJElectronics/KalmanFilter #12: i2cWrite fail : 4

haarislabs, 2018-06-06:
I sometimes get this error [error info attached in the screeshot] while reading the values of my MPU6050 sensor. I'm using an ESP8266, even after resetting the board it shows the same error, but if I unplug the the power source and put it back on then it works fine. What causes this error and how can I avoid it ???

[image]

rocheparadox, 2018-06-20:
Either your wire connection is not solid(i.e loosely fit) or your MPU6050 has failed. Elements on the module are SMDs you know. They have to be handled with care. I had the same problem with one of my MPU6050s. It would show some erratic value for sometime and then show the same error message as yours.

Lauszus, 2018-06-23:
As @rocheparadox wrote the "i2cWrite failed" warning is probably a bad connections. Please make sure that you have a soldered connecting between the IMU and ESP8266.

closed 2018-06-26 by haarislabs