TKJElectronics/KalmanFilter #17: error:  i2cWrite failed: 2 ~~~> why?

dsyleixa, 2018-09-12:
hi, I loaded your MPU6050 example to my Arduino Due
SCK, SDA, 
Vcc 3.3V
GND
A0 --> GND

the device is found by i2c scanner at addr 0x18.

Why is ths error message
 i2cWrite failed: 2
with you code?

Lauszus, 2018-09-12:
The address of the MPU-6050, which the example is written for is 0x68.

Have you added any pull-up resistors to the I2C lines?

dsyleixa, 2018-09-12:
I have a Due, there are built-in pullups.
The i2c scanner (arduino prog) detects it fine at 100k and 400k i2c clock speed.

dsyleixa, 2018-09-12:
PS,
I changed lines 57-61 in MPU6050.ini Sketch  to

```
  while (i2cRead(0x25, i2cData, 1));
  if (i2cData[0] != 0x18) { // Read "WHO_AM_I" register
    Serial.print(F("Error reading sensor"));
    while (1);
  }
```

still 
i2cWrite failed: 2

Lauszus, 2018-09-12:
That is not the place to set the I2C address. The address is set here: https://github.com/TKJElectronics/KalmanFilter/blob/ed2d15a230db4f9833ff03f2d2cf5af8eb6cd58c/examples/MPU6050/I2C.ino#L18

However this does not explain why the MPU-6050 should have an address of 0x18. Are you 100 % sure that you actually have an MPU-6050? What does it say on the IC on the breakout board?

dsyleixa, 2018-09-13:
not much is clearly detectable there actually, sth like
CJMCU-0808
I once got it by ebay from China, but the breakout board looks like pictures from MPU6050 in the web, just for 3.3V having no voltage stabilisation though.

Lauszus, 2018-09-13:
No you only need to change the address of the sensor, but again I doubt you actually have a MPU-6050, as the ID can only either be 0x68 or 0x69.

Lauszus, 2018-09-13:
Btw get a magnifying glass to read the number of the IC.

dsyleixa, 2018-09-13:
after restoring lines 57-61 and changing line 18 to 0x18, there is a new error now: 3 instead of 2:

i2cWrite failed: 3

Lauszus, 2018-09-13:
No it uses 0x68 and 0x69 as well. Here is a list of known IDs by Adafruit: https://learn.adafruit.com/i2c-addresses/the-list#0x18-2-5.

dsyleixa, 2018-09-13:
ok, then I'm afraid I have to resign, thank you!

Lauszus, 2018-09-13:
No worries :)

closed 2018-09-13 by Lauszus

dsyleixa, 2018-09-13:
update: I ordered a new one, looking forward... :)

Lauszus, 2018-09-13:
Nice. Feel free to reopen the issue if you have problems with the new one as well.

Bringamosa, 2019-03-14:
Probably to late but, in your sketch folder you will need the I2C.ino file to get rid of that error.

I need in total 4 files go get it to work.
MySketch.ino
I2C.ino
Kalman.h
Kalman.cpp