TKJElectronics/KalmanFilter #19: how to retrieve relative heading = yaw ?

dsyleixa, 2018-09-18:
hi
how is it possible to retrieve relative heading = yaw ?

dsyleixa, 2018-09-26:
will there be a way to get yaw here in the future?

dsyleixa, 2018-10-02:
is this repository no longer maintained for issues and requests?

Lauszus, 2018-10-02:
No not with this particular code, but you could follow the same principals as described in the blog post: http://blog.tkjelectronics.dk/2012/09/a-practical-approach-to-kalman-filter-and-how-to-implement-it/ and then apply it to your own application. That's the beauty of the Kalman filter.

Anyway check out the code for my flight controller, as calculate the yaw there as well. The relevant code is here: https://github.com/Lauszus/LaunchPadFlightController/blob/master/src/IMU.c.

closed 2018-10-02 by Lauszus

dsyleixa, 2018-10-03:
thanks, I checked that but con't find any relevant code snippets.
Can you please copy and paste them?

dsyleixa, 2018-10-03:
thanks, I checked that but can't find any relevant code snippets about yaw, and the Kalman maths is far beyond my skills.
Can you please copy and paste the function code  from your  flight controller?

BTW, as the issue is not resolved yet, would you please reopen this issue? Finally it's an enhancement request.