TKJElectronics/KalmanFilter #20: roll angle computed in inverse direction by this MPU6050 lib?

dsyleixa, 2018-09-21:
hi,
As I just observed, your roll angle is computed the inverse direction by this MPU6050 lib.
I compared your values with the values I get from my CMPS11 IMU, and also compared it by this chart:
https://i.stack.imgur.com/ge9sN.png
accx  to those references,
rolling clockwise  (to the left) it's negative, 
whilst rolling counter-clockwise (to the right) it's positive there.

By your MPU6050 example it's other way round as far as I see.

Lauszus, 2018-10-02:
@dsyleixa you're welcome to open up a PR with the fix.

dsyleixa, 2018-10-03:
I have no idea how a PR works, I just use github to download files and libs. But eventually it's not already verified about this issue or failure, I just came upon that by comparing 3 different code references: so who is right or wrong?