TKJElectronics/KalmanFilter #23: Full range with MPU

Pawlowski123, 2019-06-13:
Hi guys,

First time I'm interacting at GitHub so sorry for any mistake.

I read the PDF that the lib is based and understood that it's not possible to use it for a 360 degrees range. I'm intrigued beucase I'm using MPU6050 with some trigonometry (atan2) and complementary filter and I'm able to use it for the full range and the Z-axis as well (of course, the closer to perpendicularity to the floor, worst or cant get the result).

Is there a way to use this library in that way? It really helped me.

mindThomas, 2019-06-13:
This is due to the inherent singularity that is present when using Euler angles, e.g. Roll angle.
If you would like to get full 360 degree singularity-free estimates around all axis, one way to do so would be with Quaternions. I would recommend you to take a look at the Madgwick filter (https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/) which is available for the MPU6050 as well here: https://github.com/kriswiner/MPU6050
Another option, if you just need 360 degree estimates for a single axis, would be to do proper wrapping and unwrapping. In that case you should still be able to do it with your current complimentary filter implementation.

Pawlowski123, 2019-06-25:
I'll try the Madgwick filter lib and post here as feedback. Thanks!

ibrahimqazi, 2025-08-07:
@Pawlowski123 Hello, do you have any feedback to share. Thank you