TKJElectronics/KalmanFilter #28: Just asking if its just angular position or more.

tugrul512bit, 2023-02-09:
Does this filter take derivative of angle and second derivative of angle  into consideration? I guess its only angular position and angular velocity because the matrix is 2x2? If not, what kind of changes would be needed to add velocity & acceleration too?

mindThomas, 2023-02-09:
You are correct, this the estimated states of this filter only includes position and thus it takes in angular velocity as an input to the propagation model.
By extending the state space to include velocity you would have a dynamic model which takes in angular acceleration. Do you have a sensor to measure that or what other idea would you have to drive the model?

mindThomas, 2023-02-09:
> backward-discrete-derivative approach (using just the position & timestep
Yes, that would just be the numerical derivative which would be very noisy. So there would not be much a model-based Kalman filter could do to improve this.
The Kalman filter can be designed to act as a low-pass filter on these numerical derivates to give you a smoothed velocity estimate, if that is what you are looking for? One way to do this is to design the model as a Constant-velocity model with only noise on the inputs. In this case your measurement vector would include both the position measurement and the angular velocity measurement.

tugrul512bit, 2023-02-09:
My intention is just to filter noise from signal without delay. Many algorithms add delay either by computational steps requiring an output delay or just the computations themselves give a phase delay. To overcome this, prediction is required. Kalman can do this right? Maybe even with just angular position?

mindThomas, 2023-02-09:
You are right. Both FIR and IIR filters will introduce a lag in real-time settings. If you don't need this at real-time (closed loop applications) you can do zero-phase filtering by running the filters forward and backward.

The Kalman filter in our case greatly reduces the lag for the angular position since it is model-based, driven by the derivative of the state to estimate, namely the angular velocity.
A Kalman filter can however not achieve zero lag since the Kalman Filter, using the estimated or tuned Measurement and Process Noises, balances between being very adaptive with short lag but a large error variance in steady state to being an aggressive smoother with long lag but a small error variance in steady state.

Since you want angular velocity without any delay you would need the derivative of this as an measured input.

tugrul512bit, 2023-02-09:
Thank you for all the explanations. Have a nice day.

closed 2023-02-09 by tugrul512bit